brenocq/atta #6: Project doesn't install anything

yurivict, 2021-03-11:
Please add the ```install``` target.

Brenocq, 2021-03-26:
Thank you, I just updated the atta AUR package to generate the atta executable.

yurivict, 2021-03-26:
But installation is supposed to be done 100% from inside of the project. The atta AUR port shouldn't run the ```install``` command.

Brenocq, 2021-03-26:
Oh, thanks for letting me know, I really appreciate it. Sorry about that, I just started with cmake and AUR packages... I'm going to reopen this issue and study more about it.
I recreated the cmake file and the pkgbuild file in the past few days, if you find something strange there, please tell me.
Thanks!

Brenocq, 2021-09-06:
Hi, now the project is installing the `atta` and `atta_test` executables. I'm closing this issue, if you find anything wrong with the new build, please let me know. Thank you!

closed 2021-09-06 by Brenocq