brenocq/atta #7: hide tab bar doesn't actually hide the tab

diksown, 2021-11-18:
Before:
[image]

After:
[image]

brenocq, 2021-12-01:
Thank you for submitting the issue! On the following days I will redo the UI windows to not show the "Hide tab bar" button.

brenocq, 2021-12-04:
The "Hide tab bar" menu was removed in commit 0ad61f15d51dc68fd176a35aa70d3d871db8e72d, thanks again for your feedback 😃 
[image]

closed 2021-12-04 by brenocq