brenocq/atta #60: ROS Integration

brenocq, 2025-01-16:
**Feature description**
For Atta to be actually usable in the robotics community, ROS integration is a must.

**Tasks**
- [ ] Study ROS documentation
- [ ] Plan ROS integration
- [ ] Implement ROS integration

ahzary, 2025-02-06:
@brenocq i would be glad to work on this.

brenocq, 2025-02-10:
Hi @ahzary! It would be amazing to get some help developing atta. Feel free to reach out to me by email (it's in my GitHub profile) to discuss about the ROS integration :)

I haven't given much thought to this integration yet, so some initial planning to think about how the user would interact with ROS is still needed.

ahzary, 2025-06-01:
[image]
last commit now send tf data properly regarding parent-child relation.
if possible can we restrict the user naming in atta to not use special characters except space and underscore?