brenocq/atta #66: Vulkan error in `dev` branch

brenocq, 2025-03-02:
**Describe the bug**
When trying to build the `dev` branch in Ubuntu 24.04, I got the following error:

```bash
[gfx::vk::Shader] Failed to compile shader shaders/compute/entityClick.asl:
sh: 1: glslc: not found

[gfx::vk::Shader] Failed to compile shader shaders/pbrRenderer/pbrRenderer.asl:
sh: 1: glslc: not found

[gfx::vk::Shader] Failed to compile shader shaders/grid/grid.asl:
sh: 1: glslc: not found

[gfx::Manager] Create mesh atta::gfx::GridPipeline[0]
[gfx::vk::Shader] Failed to compile shader shaders/drawer/line.asl:
sh: 1: glslc: not found

[gfx::vk::Shader] Failed to compile shader shaders/drawer/point.asl:
sh: 1: glslc: not found

[Atta] Initialized
[gfx::Manager] Create mesh atta::gfx::Drawer::line
[gfx::Manager] Create mesh atta::gfx::Drawer::point
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::RenderQueue] Failed to submit render queue!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::vk::Fence] Failed to wait for fence!
[gfx::VulkanAPI] Failed to submit draw command buffer!
[Atta] Finished

```

**Expected behavior**
There should be no Vulkan problems when running on Ubuntu. If Vulkan is not available, OpenGL should be used instead.

brenocq, 2025-03-02:
There appear to also have some errors when running atta with MacOS (building works fine, but there is a GLFW error when running)
```bash
System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Codes:       0x0000000000000001, 0x0000000000000020

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [29627]

VM Region Info: 0x20 is not in any region.  Bytes before following region: 4374872032
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104c34000-1050f8000    [ 4880K] r-x/r-x SM=COW  /Users/USER/*/atta

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   atta                          	       0x104c40664 glfwSetWindowUserPointer + 16
1   atta                          	       0x104fee984 atta::graphics::GlfwWindow::GlfwWindow(atta::graphics::Window::CreateInfo const&) + 152
2   atta                          	       0x104fe58b0 atta::graphics::Manager::startUpImpl() + 208
3   atta                          	       0x104f76444 atta::Atta::Atta(atta::Atta::CreateInfo const&) + 112
4   atta                          	       0x104c38bb0 main + 724
5   dyld                          	       0x1982e0274 start + 2840
```

ahzary, 2025-03-02:
this is the error encountered when building the dev branch on ubuntu 22.04,

[image]

brenocq, 2025-03-05:
Hi @ahzary! I've temporarily disabled Vulkan in the `dev` branch and fixed the OpenGL errors in PR #70. You should be able to run `dev` without graphics errors now, let me know if you still have issues :)

closed 2025-03-10 by brenocq